microsoft/uf2-samdx1 #22: Network profiles showing up on Mac

microsoftsam, 2017-06-01:
Circuit Playground / Arduino Zero boards are showing up in as a Network profile on my Network settings on a Mac. 
It's unclear to me when they're created, but I have a good odd 50 of them. It seems to me that it thinks the device is a Modem.

[image]
[image]
[image]
[image]

ladyada, 2017-06-01:
maybe a serial port thing? on windows we get multiple COM ports, they're not harmful but they exist :)

jamesadevine, 2017-06-01:
Yup! Pretty sure this is Serial ports :) I have at least a trillion entries for CMSIS-DAP (micro:bit)

Not sure there is anything that can be done, as @ladyada said, they're not harmful - but they are spurious.

James.

mmoskal, 2017-06-05:
Attached is a node.js script which removes all `usbmodem` network interfaces. 

[clean-ifaces.zip](https://github.com/Microsoft/uf2-samd21/files/1051833/clean-ifaces.zip)

I believe new entries are created when we change USB ID or in similar circumstances, so regular users are unlikely to accumulate 170 of these like I did. As it seems the same thing happens on microbit/DAPLink and also on the AVR bootloader, I don't think we can do much about it.

closed 2017-06-05 by mmoskal