microsoft/uf2-samdx1 #25: CDC with MSC on Windows

dhalbert, 2017-09-11:
@mmoskal Hi, I'm debugging some issues with the Adafruit board drivers on Windows 7 and UF2. For some people, installing the drivers prevents the UF2 MSC `xBOOT` drives from appearing when the bootloader is enabled. For other people (e.g., @ladyada) they still see the `xBOOT` drives even with the drivers installed. I've duplicated the problem on a fresh Win7 installation, but it doesn't show up completely consistently.

The [README.md](https://github.com/Microsoft/uf2-samd21/blob/master/README.md) says using CDC and MSC together could be an issue:

> ... On Windows, if you have drivers installed for the USB ID chosen, then CDC might work and MSC will not work; otherwise, if you have no drivers, MSC will work, and CDC will work on Windows 10 only. Thus, it's best to set the USB ID to one for which there are no drivers.

When the drives don't appear, I notice that the composite USB device does not show up, only the CDC device (using Device Manager or http://www.uwe-sieber.de/usbtreeview_e.html to check this). I though this might be due to `&MI=0` present or missing for the particular USB PID in the driver `.inf` file, but that doesn't seem to be it.

Could you expand on your comment above? Do you know the cause of the issue with respect to the drivers? Thanks!

mmoskal, 2017-09-13:
Unfortunately, I'm not much of a Windows driver expert. I thought things were working, as reported by @ladyada, but it seems they might not. It's clearly possible to make it all work, as evidenced by ARM MBED drivers, which support both MSC and CDC (and HID), but I'm not quite sure how.

ladyada, 2017-09-13:
it does work but only if you do a golden path install. deviations dont work. we're not sure whts up but somehow it seems (i think) that the win7 os attaches the CDC driver to the MSC endpoints!

dhalbert, 2017-10-07:
We fixed this by always specifying `MI` in the device descriptors in the `.inf` file. See https://github.com/adafruit/Adafruit_Windows_Drivers/pull/5.
So now we have entries like
```
"%METRO_M0_EXPRESS% (0013:00)"=DriverInstall, USB\VID_239A&PID_0013&MI_00
"%METRO_M0_EXPRESS% (8013:00)"=DriverInstall, USB\VID_239A&PID_8013&MI_00
"%METRO_M0_EXPRESS% (8014:00)"=DriverInstall, USB\VID_239A&PID_8014&MI_00
```
Previously, for some drivers, we only had something like:
`USB\VID_239A&PID_0013`
Composite devices that did not specify MI=00 were being installed and then were hiding and/or confusing the multiple interfaces presented by the devices. So for instance, the MSC device for UF2 was not appearing.

closed 2017-10-07 by dhalbert