microsoft/uf2-samdx1 #27: Can't set the fuses

deshipu, 2017-10-20:
The `scripts/dbgtool.js fuses` command fails for about half of the chips that I have with:

```
$ scripts/dbgtool.js fuses
File: fuses
Starting /home/sheep/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino/bin/openocd -d2 -s /home/sheep/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino//share/openocd/scripts/ -f /home/sheep/.arduino15/packages/arduino/../adafruit/hardware/samd/1.0.19/variants/trinket_m0/openocd_scripts/arduino_zero.cfg -f scripts/fuses.tcl
Open On-Chip Debugger 0.9.0 (2015-09-02-10:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.463901
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
Fuses:
d8e0c7fa
fffffc5d
*** Updating fuses! ***
d8e0c7ff
fffffc5d
Error: array2mem: Write @ 0x41004000, w=2, cnt=1, failed
/home/sheep/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino//share/openocd/scripts//memory.tcl:122: Error: memwrite16: 
in procedure 'script' 
at file "embedded:startup.tcl", line 60
in procedure 'cmd' called at file "scripts/fuses.tcl", line 50
in procedure 'memwrite16' called at file "scripts/fuses.tcl", line 24
at file "/home/sheep/.arduino15/packages/arduino/tools/openocd/0.9.0-arduino//share/openocd/scripts//memory.tcl", line 122
```

It seems like they are somehow locked, but doing `chip-erase` on them doesn't seem to help.

mmoskal, 2017-10-21:
Could you try IBDAP? I've never used stlink.

deshipu, 2017-10-21:
I only tried with stlinkv2 and j-link, I don't have ibdap, but I thought that the protocol is the same no matter what programmer is used?

mmoskal, 2017-10-21:
It should be. I understand it works on some of your chips? Try comparing log output.

My suspicion about jlink was the frequency - I think it should be 1 MHz. But I'm not really much of a SWD expert.

mmoskal, 2017-12-18:
Closing due to no further input.

closed 2017-12-18 by mmoskal

deshipu, 2017-12-18:
I have no further information to provide, it doesn't mean that the problem is solved, though.

mmoskal, 2017-12-18:
so there's no interesting difference in openocd logs between chips that work and ones that don't?

deshipu, 2017-12-18:
No, except for that moment where it fails writing because it's protected.

mmoskal, 2017-12-18:
I think Atmel studio can set these fuses. You can check if it works. @ladyada have you ever seen something like this?

deshipu, 2017-12-18:
Unfortunately I don't have access to a Windows machine to try Atmel Studio right now, I will try when I get an occasion.

ladyada, 2017-12-18:
ya, openocd is flakey and not well supported for samd21, we recommend jlink.exe or atmel studio only!

deshipu, 2017-12-21:
I now confirmed that all of those chips that didn't work with openocd can be programmed with atmel studio. I'm sorry for the noise.