microsoft/uf2-samdx1 #28: compiled and flashed UF2 for Arduino Zero, no go

tballmsft, 2017-10-25:
I followed directions, built the binary for UF2 for Zero, successfully flashed it, but afterwards Zero doesn't show up as a drive.

tballmsft, 2017-10-25:
I built on Windows... Richard is trying on Unix.

tballmsft, 2017-10-25:
Sam built on MacOS and I tried his INO. still same issue - flashed and verified, but no MSC

tballmsft, 2017-10-25:
Sam flashed and verified on MacOS - same issue there - no MSC.

tballmsft, 2017-10-25:
comparison point: I was successful with MKR1000.

tballmsft, 2017-10-25:
OOPS. I was programming the debug chip... oh well. It worked on the native. Live and learn...

closed 2017-10-25 by tballmsft

mmoskal, 2017-10-26:
How did you program the wrong chip? Was it using openocd? Or arduino sketch?

microsoftsam, 2017-10-26:
Using Arduino Sketch. The Zero has two USB ports. It's important to use the right one. Both @tballmsft and I made the same mistake :P

mmoskal, 2017-10-26:
It's actually possible to program zero through both, but you have to choose different programmer.

Maybe the instructions should say: first make sure blinky works?