microsoft/uf2-samdx1 #29: Compiler flags

awatterott, 2017-12-18:
No long-calls will reduce the code size. Is there a reason why `-mlong-calls` is set?

`-nostartfiles` can also be set, because `startup_samd21.c` includes everything. This will also reduce the size.

mmoskal, 2017-12-18:
Thank you, it saves us 400 bytes! I'm not sure where it came from, must have been in the original Atmel example.

closed 2017-12-18 by mmoskal

ladyada, 2017-12-18:
wow that's great - @mmoskal anything we wanted to add? im doing an update to the tester/programmer so now is a good time to tweak UF2

awatterott, 2017-12-18:
We are working on binary file support, which is working but there seems to be some general USB problems. I have opened an [issue](https://github.com/Microsoft/uf2-samd21/issues/30) for it.

mmoskal, 2017-12-18:
@ladyada I'm working on handover from application to bootloader, so we can flash from PXT without tapping reset. It is working for me (with my most recent checkins) but if you give me two more days, I can test it a little bit better.