microsoft/uf2-samdx1 #30: USB problems

awatterott, 2017-12-18:
We are currently testing the bootloader, but had some problems. For example the USB Descriptor Strings are not correct and if we access the USB device under Windows with the [USB Device Tree Viewer](http://www.uwe-sieber.de/usbtreeview_e.html) then the USB mass storage device will not work after the access.

For the string problem, I think it is the transmitted length here:
https://github.com/Microsoft/uf2-samd21/blob/master/src/cdc_enumerate.c#L779
`sendCtrl(&desc, sizeof(StringDescriptor));`
I think it should be
`sendCtrl(&desc, desc.len);`

mmoskal, 2017-12-18:
Does it fix the issue for you? There's a separate limit on transmission length in sendCtrl(), but probably not related.

awatterott, 2017-12-18:
The modification fixes the string transmission problem. But there is still another problems, which can be reproduced with the USB Device Tree Viewer.

awatterott, 2017-12-18:
@mmoskal Could you reproduce the issue?
We have made also a [CDC+MSC bootloader](https://github.com/watterott/SAM-BAR), which is based directly on ASF and there we have no problems, but it is also much bigger. On our bootloader I have also not seen any limitations with CDC+MSC with or without installed drivers on a Windows OS.

mmoskal, 2017-12-18:
Unfortunately I have no Windows machine to test it on right now...

mmoskal, 2017-12-20:
@awatterott it's possible it's related to this:

https://stackoverflow.com/questions/3739901/when-do-usb-hosts-require-a-zero-length-in-packet-at-the-end-of-a-control-read-t

I just did a fix in Codal. You can try to do something similar in bootloader and see if it works: https://github.com/lancaster-university/codal-samd21/commit/ed3d6de7c8836a4248a610e2ac986d807907746a

awatterott, 2017-12-20:
I tried different things, but I dont find a cause for the issue.
Could you reproduce the behavior?
I can make a pcap log for wireshark, if it helps?

awatterott, 2017-12-21:
I made some progress. It has to do with the descriptor strings. If I remove all strings from the device descriptor then there are no issues.

awatterott, 2017-12-21:
The issue appears when the host tries to read string descriptors that do not exist. Under Windows for example the string with index [0xEE](https://msdn.microsoft.com/en-us/library/windows/hardware/ff537430(v=vs.85).aspx). Currently the [STALL](https://github.com/Microsoft/uf2-samd21/blob/master/src/cdc_enumerate.c#L765) sending seems to make the problems.
Here is a workaround, which I have tested on different Windows machines:
https://github.com/awatterott/uf2-samd21/commit/48df8c2ac9bd97a534c6a76454e0141169d32677

awatterott, 2018-02-11:
I have reviewed the USB issues with a hardware analyzer and the problem is, that a wrong string descriptor is sent after STALL.
https://github.com/awatterott/uf2-samd21/commit/279da2a3824d44bf8e4d0f35321cb2b68dad37eb
Should I create a PR for this.

awatterott, 2018-02-19:
@mmoskal Did you have time to look into this?  I think it is important to fix the bug, because it will bring the device to freeze/hang.

pelikhan, 2018-02-19:
@awatterott could you prepare a PR?

awatterott, 2018-02-19:
Sure, it is PR https://github.com/Microsoft/uf2-samd21/pull/33

closed 2018-02-21 by awatterott