microsoft/uf2-samdx1 #38: Single Button Press into Bootloader

brendanvanbreda, 2018-03-20:
I am attempting to build a custom binary release for a custom board, with [reference](https://makecode.com/blog/one-chip-to-flash-them-all) about the reset procedure.

When looking through [uf2.h](https://github.com/Microsoft/uf2-samd21/blob/master/inc/uf2.h)
```
// If enabled, bootloader will start on power-on and every reset. A second reset
// will start the app. This only happens if the app says it wants that (see SINGLE_RESET() below).
// If disabled here or by the app, the bootloader will only start with double-click of the reset
// button.
#define USE_SINGLE_RESET 1
```

If this value is set to 0, should the expected result be that the bootloader will become active on a single reset ?

As when working in [maker.makecode](https://maker.makecode.com/) with the Arduino Zero it makes use of a single reset to enable the bootloader. In comparison to when converting a .bin file from Arduino to a .UF2 it would still require a second reset. This occures between uploads of the UF2 file from the different sources.

mmoskal, 2018-03-20:
To get single-reset behavior on Arduino-generated binaries you will need to set USE_SINGLE_RESET to 1, and also modify:
```
#define SINGLE_RESET() (*((uint32_t *)0x20B4) == 0x87eeb07c)
```
to
```
#define SINGLE_RESET() 1
```
Let me know if it didn't work!

closed 2018-03-20 by mmoskal

brendanvanbreda, 2018-03-22:
Awesome thanx @mmoskal  worked as expected.

TamojitSaha, 2018-12-21:
Is it possible to have a key combination to enter into bootloader mode?

mmoskal, 2019-01-04:
You could probably quite easily modify the bootloader to support that - just configure the pin as input with a pull up (or pulldown, depending on how your button is wired) and return at the beginning of `check_start_application()` when your button is pressed.