microsoft/uf2-samdx1 #50: Linker error - section .binfo LMA overlaps section .relocate LMA

ademuri, 2018-06-30:
I'm seeing issue #24 again. I'm using GCC 6.3.1, as suggested in the bug:

`$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

And get this error:

`arm-none-eabi-gcc -Lbuild/zero -mthumb -mcpu=cortex-m0plus -Os -g -DSAMD21 -Wall -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols -save-temps -nostartfiles --specs=nano.specs --specs=nosys.specs \
         -Tscripts/samd21j18a.ld \
         -Wl,-Map,build/zero/bootloader-zero-v2.0.0-23-gc92ac1d.map -o build/zero/bootloader-zero-v2.0.0-23-gc92ac1d.elf build/zero/flash_samd21.o build/zero/init_samd21.o build/zero/startup_samd21.o build/zero/usart_sam_ba.o build/zero/utils.o build/zero/cdc_enumerate.o build/zero/fat.o build/zero/main.o build/zero/msc.o build/zero/sam_ba_monitor.o build/zero/uart_driver.o build/zero/hid.o
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: section .binfo LMA [0000000000001ff0,0000000000001fff] overlaps section .relocate LMA [0000000000001e60,000000000000202b]
collect2: error: ld returned 1 exit status`

The weird thing is that this used to work, and when I upgraded from Ubuntu 17.10 to 18.04, it broke. It looks like I was [probably using](https://packages.ubuntu.com/artful/gcc-arm-none-eabi) GCC 5.4.1 before (that's the current version for 17.10).

dhalbert, 2018-06-30:
We (Adafruit, not MS) have been building with gcc 7.2.1 from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads (2017q4). There was no package for bionic (18.04) in the ppa, but the xenial one works ok, or you can download the tar.gz and add it to your path manually.

2018q2 _just_ was released. I haven't tried that.

ademuri, 2018-06-30:
Ah ok, using that particular version of the toolchain worked. Thanks for the help!

closed 2018-06-30 by ademuri