microsoft/uf2-samdx1 #53: MKR1300

sabas1080, 2018-07-27:
Hi
I'm trying to walk this on the MKR 1300, but when trying to load via USB Storage the transfer of the uf2 file never ends

and a sketch with Arduino the next error:

```
SAM-BA operation failed
Set binary mode
readWord(addr=0)=0x20002dd8
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
version()=v1.1 [Arduino:XYZ] Jul 26 2018 20:24:37
chipId=0x10010005
Connected at 921600 baud
readWord(addr=0)=0x20002dd8
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMD21G18A
readWord(addr=0)=0x20002dd8
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010305
Chip ID      : 10010005
version()=v1.1 [Arduino:XYZ] Jul 26 2018 20:24:37
Version      : v1.1 [Arduino:XYZ] Jul 26 2018 20:24:37
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 5.005 seconds

Write 18800 bytes to flash (294 pages)
write(addr=0x20005000,size=0x1000)
Ha ocurrido un error mientras se enviaba el sketch
```

Thanks

mmoskal, 2018-07-27:
Did you try on a different SAMD21 board with the same bootloader? I don't think MKR1300 should be particularly different from all the other boards...

sabas1080, 2018-07-27:
Yes, I'm try with SAMD Sparkfun Dev

mmoskal, 2018-07-27:
Maybe you have a defective MKR1300 board?

sabas1080, 2018-07-28:
With SAMBA working

sabas1080, 2018-07-30:
My board defective, thanks

closed 2018-07-30 by sabas1080