microsoft/uf2-samdx1 #55: SAMD51 firmware update dual banks

WalterPrech, 2018-08-02:
Hi,
I read, that the SAMD51 have the possibility to do firmware updates with writing the new firmware to BankA or BankB and then switching to the bank, containing the new firmware.

How to do this? Must the bootloader configuration be changed? Do you have an example for this?

mmoskal, 2018-08-03:
I don't think this is supported. @dhalbert ?

dhalbert, 2018-08-03:
I see there is such a feature in the SAMD51. But we wouldn't bother with it for a couple of reasons:
1. Our firmware loads are larger than half the NVM, so we can't use this feature.
2. Since we have a protected bootloader, a bad load can easily be recovered from.

closed 2018-08-14 by WalterPrech

CraigHutchinson, 2020-06-15:
I am looking into this dual-bank mechanism and I don't stand to fully understand.

Naively I would agree with @dhalbert in respect to it halving the usable flash space but I cannot actually see any confirmation that this is how it functions when  active i.e. It appears the entire flash is dual banked and I cant see any reference in the Datasheet to say you enable it and get half the space. 

If anybody can confirm or find a reference this will be handy!

dhalbert, 2020-06-15:
I agree it is poorly explained in the data sheet. There is no explicit description that BANKA and BANKB  are each half of the main part of flash. But there are hints in the SmartEEPROM description. And it seems very unlikely that the flash capacity would actually be double what was spec'd and so much chip area would be devoted to implementing this feature.

ricardoboss, 2022-05-30:
What about `SAMD51J20A`? It would have 512K flash per bank. Is support for this planned?

ricardoboss, 2022-05-30:
Ref: https://microchipsupport.force.com/s/article/SAM-D5x-E5x---NVM-Bank-Swapping