microsoft/uf2-samdx1 #56: Some disk monitoring programs have trouble with UF2 FAT filesystem

dhalbert, 2018-08-23:
Several people have discovered that running programs like [AIDA64](https://www.aida64.com/) and [Hard Disk Sentinel](https://www.hdsentinel.com/) cause Windows Explorer to hang when trying to read or write files from the UF2 "filesystem".

Some reports:
https://forums.adafruit.com/viewtopic.php?f=60&t=135400
https://forums.adafruit.com/viewtopic.php?f=52&t=140056
https://forums.adafruit.com/viewtopic.php?f=52&t=140028

I've reported this to AIDA64: https://forums.aida64.com/topic/4296-uf2-bootloader-usb-drive-and-aida64-cause-explorer-to-freeze/. They will investigate

I'm not sure where to place responsibility for this, but perhaps there's some extra emulation that might help. I don't think it's good to cause a freeze-up though, if a drive is not acting according to expectation.

dhalbert, 2018-09-17:
AIDA64 has a beta that is supposed to fix this: https://forums.aida64.com/topic/4296-uf2-bootloader-usb-drive-and-aida64-cause-explorer-to-freeze/?do=findComment&comment=20455

ladyada, 2018-09-17:
neat!