microsoft/uf2-samdx1 #57: SAMD21, update bootloader with security bit set

fedorcomander, 2018-08-30:
Hi Guys!

Does update bootloader functionality work with security bit set???
I got mine update firmware looping at setBootProt(7); in selfmain.c

So, clearly bootprot was not changed from 2 to 7 and app restarts.

Help!
Alex

ericbaril72, 2018-08-30:
The "Make" process for the bootloader-updater creates a specific .bin using sketch.cpp.

sketch.cpp DOES change the security bits BEFORE and AFTER updating the bootloader and finishes by "erasing itself" ( harakiri ).  This will keep the device in Bootloader mode after reset since no App is present at START_ADDRESS.

Assuming you started with a Bossac bootloader, and upload the "bootloader-updater" sketch .bin file with your standard Arduino IDE "upload button", you will now have a UF2 bootloader.

If you don't see a new Drive showing up in your windows explorer, Go into your "Device Manager" and Erase the Driver that shows-up in USB for your arduino.  Replugging your device will re-install the driver and most likely detect the "USB-MSC".

fedorcomander, 2018-08-30:
I don't use sketch and arduino and I am already have uf2 bootloader loaded on the samd21, security bit set and BOOTPROT fuse set to 2. So far so good.

Now i want to update bootloader with new version. I am uploading generated by make script update-bootloader-my-board.uf2... (which is compiled version of selfmain.c + selfdata.c)... 

Existing uf2 bootloader loads it in flash and starts it as usual from 0x2000....

Now, selfmain.c code gets stuck on changing BOOTPROT fuse back to 7, and app restarts.

It doesnt execute any further then

   #ifdef SAMD21
    setBootProt(7); // 0k
    #endif

Which makes sense, only if BOTTPROT is not being changet do 7...

Please advice.

fedorcomander, 2018-08-30:
What i also see, sketch.cpp does

   NVIC_SystemReset();

and selfmain.c

resetIntoApp();

at the end of setBootProt() function....

fedorcomander, 2018-08-30:
Works fine if no security bit set btw.

fedorcomander, 2018-08-30:
With security bit set, CPU is able to write flash... I just cant set BOOTPROT fuse to 7 for reflashing the bootloader. I think problem is in auxilary space, it cant be changed with SSB set.... And thats where fuses live. Bummer.

dhalbert, 2018-08-30:
Yes, you're right. I was over-interpreting. I don't really see that in the datasheet, but it's mentioned here: https://gist.github.com/cmaglie/c411621b8d49494faeb7869ad59aa6b6.

Would you close the issue if you're satisfied? Thanks.

fedorcomander, 2018-08-30:
Yeah, its a chip limitation, not a bootloader.

Thanks for your time guys.

closed 2018-08-30 by fedorcomander