microsoft/uf2-samdx1 #59: Can't attach more than 2 devices in bootloader mode

PTS93, 2018-11-06:
I have an odd issue.
I'm using multiple Adafruit Metro M0 running an unmodified `metro_m0_0-v2.0.0-adafruit.5-4-g0a2294c`
Whenever I connect 2 Metros and put them in bootloader mode. A third or fourth Metro will not properly enumerate anymore. When I put the third in bootloader mode it just starts to flash rapidly. No MSC device popping up and no serial device either, as if it isn't connected to a computer at all.
Running 'lsusb' shows at least that USB enumeration worked and shows the VID:PID for that Metro (I assigned each metro their own PID for testing purposes).

I'm running Ubuntu 18.04.

PTS93, 2018-11-06:
After playing a while with it I noticed it is not a hard limit. I even got to the point where no Metro is showing up anymore and all flashing rapidly. After a reboot all Metros are good again.
I feel like it correlates at least somehow to how often Metros connected as bootloaders but I'm not sure on that. So far it seems really random.

mmoskal, 2018-11-06:
I believe they all use the same serial number. That might be a problem in this scenario.

PTS93, 2018-11-06:
serialNumber and pnpId are different for each
I haven't checked but I think the serial number comes from the 128bit uuid of the samd.

dhalbert, 2018-11-06:
`/var/log/syslog` will show info about what the MSC device is reporting, including serial numbers, and might also show indications of what's going wrong on the boards that aren't showing up properly.

PTS93, 2018-11-06:
Thank @dhalbert that was a good hint.
Whenever a Metro goes havoc I get the messages below.

```
Nov  6 19:32:45 server kernel: [  289.292101] usb 1-4.4.4: new full-speed USB device number 16 using xhci_hcd
Nov  6 19:32:45 server kernel: [  289.406897] usb 1-4.4.4: New USB device found, idVendor=239a, idProduct=1003, bcdDevice=42.01
Nov  6 19:32:45 server kernel: [  289.406898] usb 1-4.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov  6 19:32:45 server kernel: [  289.406899] usb 1-4.4.4: Product: Metro M0 3
Nov  6 19:32:45 server kernel: [  289.406900] usb 1-4.4.4: Manufacturer: Adafruit Industries
Nov  6 19:32:45 server kernel: [  289.406901] usb 1-4.4.4: SerialNumber: 8EA9E2915150323143202020FF122929
Nov  6 19:32:45 server kernel: [  289.407884] usb 1-4.4.4: Not enough bandwidth for new device state.
Nov  6 19:32:45 server kernel: [  289.407913] usb 1-4.4.4: can't set config #1, error -28
Nov  6 19:32:45 server mtp-probe: checking bus 1, device 16: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4.4"
Nov  6 19:32:45 server mtp-probe: bus: 1, device: 16 was not an MTP device
Nov  6 19:32:45 server fwupd[2453]: failed to add USB device: 239a:1003 is not supported: USB error on device 239a:1003 : Entity not found [-5]
Nov  6 19:32:45 server upowerd[1262]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4.4/1-4.4.4
```

dhalbert, 2018-11-06:
If you do a websearch for "Not enough bandwidth for new device state", it comes up in other situations with multiple USB peripherals. I don't think this is a UF2 issue. It looks like if you're using a USB3 hub plugged into a USB2 port, you might see this, or it might happen if there are multiple devices on the same USB3 controller. If this is a desktop with many USB ports, try making sure you split the boards between multiple controllers (e.g., maybe try some ports on the back and on the front of the chassis).

PTS93, 2018-11-06:
https://bugzilla.redhat.com/show_bug.cgi?id=1411604

It seems this an issue that happens with a lot of different devices.
I'm guessing the arduino core is running the virtual cdc at low speed and UF2 as full speed?

Nontheless this issue definitely originates from the Linux kernel.
I was able to "fix" it and it is fairly odd. So far I was using a USB3.0 Hub connected to a USB3.0 port. Now I attached a USB2.0 Hub to the same USB3.0 port and tadah everything works. So by downgrading the connection I don't get bandwidth issues... :)

closed 2018-11-06 by PTS93