microsoft/uf2-samdx1 #70: UF2 BOOT drives fail to handle SMART requests appropriately

dhalbert, 2019-04-16:
See https://github.com/adafruit/uf2-samdx1/issues/50. Programs that make SMART requests to UF2 BOOT drives can cause the drive to lock up, not appear, etc. I'll look at this, but if one of you knows an obvious fix, let me know. Will PR any fixes back to you.

Extensive testing by @vitorio described in https://forums.adafruit.com/viewtopic.php?f=60&t=149791&p=743111#p743089. I also duplicated this finding. Example below. Note that the drive (or the driver?) is reporting SMART support is available.

```
$ df
Filesystem     1K-blocks      Used Available Use% Mounted on
[...]
/dev/sdc            7934       513      7421   7% /media/halbert/CPLAYBOOT

$ smartctl --all -d scsi /dev/sdc
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.0-47-generic] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               CPlay Ex
Product:              press
Revision:             1.00
User Capacity:        8,192,000 bytes [8.19 MB]
Logical block size:   512 bytes
scsiModePageOffset: response length too short, resp_len=4 offset=4 bd_len=0
Device type:          disk
scsiModePageOffset: response length too short, resp_len=4 offset=4 bd_len=0
Local Time is:        Tue Apr 16 09:09:16 2019 EDT
SMART support is:     Available - device has SMART capability.
SMART support is:     Enabled
Temperature Warning:  Disabled or Not Supported

=== START OF READ SMART DATA SECTION ===
^C^C^C^C
[hangs, killed process, which became a zombie, eventually timed out]
```

dhalbert, 2019-04-16:
Trying a few USB sticks. Most respond like this:
```
$ sudo smartctl --all -d scsi /dev/sdc
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.0-47-generic] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               USB 2.0
Product:              USB Flash Drive
Revision:             1100
User Capacity:        4,057,989,120 bytes [4.05 GB]
Logical block size:   512 bytes
>> Terminate command early due to bad response to IEC mode page
A mandatory SMART command failed: exiting. To continue, add one or more '-T permissive' options.
```

A high-quality Samsung stick I have does this:
```
$ sudo smartctl --all -d scsi /dev/sdc
smartctl 6.6 2016-05-31 r4324 [x86_64-linux-4.15.0-47-generic] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Vendor:               Samsung
Product:              Flash Drive FIT
Revision:             1100
Compliance:           SPC-4
User Capacity:        32,086,425,600 bytes [32.0 GB]
Logical block size:   512 bytes
LU is fully provisioned
scsiModePageOffset: response length too short, resp_len=4 offset=4 bd_len=0
Logical Unit id:      0x00010200000608040x2020030102060804error: SCSI name string
Serial number:        AA00000000000489
Device type:          disk
scsiModePageOffset: response length too short, resp_len=4 offset=4 bd_len=0
Local Time is:        Tue Apr 16 09:47:19 2019 EDT
SMART support is:     Available - device has SMART capability.
SMART support is:     Disabled
Temperature Warning:  Disabled or Not Supported

=== START OF READ SMART DATA SECTION ===
SMART Health Status: OK
Current Drive Temperature:     0 C
Drive Trip Temperature:        0 C

Error Counter logging not supported

scsiModePageOffset: response length too short, resp_len=4 offset=4 bd_len=0
Device does not support Self Test logging
```

mmoskal, 2019-04-16:
Probably would be best to compare usb trace from the Beagle usb debugger from the cheap pendrive and CPX. 

if these smart tools work on macos i can do that

dhalbert, 2019-04-16:
@mmoskal I was just planning to do that, with the wireshark USB tracer. But if you have time to do this soon, and have the expertise in your group, I would leave it to you. We've seen a recent noticeable increase in the number of people who are hitting this problem.

mmoskal, 2019-04-16:
Sorry, on macOS all I get is
```
$ sudo smartctl --all -d scsi disk2
smartctl 7.0 2018-12-30 r4883 [Darwin 18.5.0 x86_64] (local build)
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org

disk2: Not a device of type 'scsi'
=======> VALID ARGUMENTS ARE: ata, scsi[+TYPE], nvme[,NSID], sat[,auto][,N][+TYPE], usbcypress[,X], usbjmicron[,p][,x][,N], usbprolific, usbsunplus, sntjmicron[,NSID], intelliprop,N[+TYPE], auto, test <=======

Use smartctl -h to get a usage summary

$ sudo smartctl --all  disk2
smartctl 7.0 2018-12-30 r4883 [Darwin 18.5.0 x86_64] (local build)
Copyright (C) 2002-18, Bruce Allen, Christian Franke, www.smartmontools.org

Smartctl open device: disk2 failed: Operation not supported by device
```

vitorio, 2019-04-16:
Hi @mmoskal, looks like you're using a newer `smartctl` than we were!

The [manpage](https://www.smartmontools.org/browser/trunk/smartmontools/smartctl.8.in) for `smartctl` 7.0 says a couple things: first that SCSI support under Darwin isn't implemented, and second to get SMART data from USB drives you need to install a driver: https://github.com/kasbert/OS-X-SAT-SMART-Driver

Installing that driver on OS X _may alone_ result in your UF2 BOOT drive failing to appear, since that was my experience.

dhalbert, 2019-04-16:
I have a test bootloader that I've passed to @vitorio and another user. It acts more like most USB sticks and card readers, returning a malformed response to MODE SENSE for page 0x1c, which causes `smartctl` (and presumably other SMART-aware software) to give up trying to read SMART data. If it works for the users, I'll PR it.