microsoft/uf2-samdx1 #76: Bootloader Fails to Compile with arm-none-eabi-gcc 9

IoTPanic, 2019-09-13:
Hi All!

Full disclaimer, I am using this uf2 code to help me understand and write my own boot loader, I am a n00b atm.

So I started working with this uf2 bootloader, and the first wall I hit is when I run `make` in the project directory, I get the following output-
```

dhalbert, 2019-09-13:
Which version of the ARM toolchain are you using? We compile with 7.3.1 (7-2018-q2), downloadable from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

IoTPanic, 2019-09-13:
I was using 9.2.0, so I should just downgrade instead of getting it to compile with 9.2.0? 

`__ASM volatile ("MSR msp, %0\n" : : "r" (topOfMainStack) : "sp");`

And I have a question about this line, I see it's using the `MSR` instruction to move a value into the `msp` register to set the stack pointer, I am unfamiliar with inline assembly, I see that the %0 is passing in `topOfMainStack` without the write only operand, but than what is `"sp"` doing? I do not see another input in the instruction.

IoTPanic, 2019-09-13:
Removed the `: "sp"` and now I have a successful compile. I am awaiting response from my last comment as to what it was doing in the instruction, I don't want to make a small assembly change that will cause a issue down the road.

IoTPanic, 2019-09-13:
Ah, okay, apologies for the noobishnes, I see at the end of the extended in line assembly you put together the clobbered register list.

closed 2019-09-13 by IoTPanic

modmatic, 2020-02-19:
@IoTPanic Could you explain your last comment? Is it fine to just remove the `: "sp"`? If feels wrong to fix it that way...

IoTPanic, 2020-02-19:
@modmatic Removing `: "sp"` will allow it to compile with the version of the ARM tool chain that ships with most package managers and I have tested this fix on SAMD21 and SAMD51 processors for months with no issues. My understanding is it's just a label for the compiler, and making the change does not functionally change the clobbered list.

cozzyd, 2020-05-20:
One can also just add -Wno-deprecated to the WFLAGS in the Makefile

dhalbert, 2020-05-20:
I have a fix for this in the adafruit fork and will be doing a PR relatively soon.

electronrancher, 2020-06-03:
@dhalbert could you please point me to your fix for this error?  I am comparing your adafruit fork and I don't see any makefile edits or other changes that look promising.  thanks

electronrancher, 2020-06-03:
Thanks @cozzyd updating the WFLAGS worked for me with latest binutils

dhalbert, 2020-06-03:
@electronrancher Change is here:
https://github.com/adafruit/uf2-samdx1/blob/246e826e910f17bf474047bce74f21b87a082b6b/lib/cmsis/CMSIS/Include/core_cmFunc.h#L456-L457

electronrancher, 2020-06-03:
Thanks @dhalbert