microsoft/uf2-samdx1 #78: Building for SAMD51 without UART

wallarug, 2019-10-08:
This is a query - rather than an issue.

How do you build a SAMD51 bootloader without the USART functionality?

I tried just removing all the arguments but it looks like they get re-added in anyway,

Any assistance would be great.

Thanks in advance.

dhalbert, 2019-10-08:
Nope, we haven't tried this, since the 16kB allocated for the SAMD51 bootloader means we haven't had to try to squeeze it to save space.

wallarug, 2019-10-08:
So does that mean it is not implemented / cannot be done at this very moment in the same way as SAMD21?

dhalbert, 2019-10-08:
No, it just means I haven't tried it :). Sounds like a bug.

wallarug, 2019-10-08:
😎 

Righto - well when you do try - this is what happens...

[image]

It complains that:

* BOOT_GCLK_ID_CORE
* BOOT_GCLK_ID_SLOW
* BOOT_USART_BUS_CLOCK_INDEX 

... are not defined.

Interesting that it doesn't complain about the other ones.

dhalbert, 2019-10-08:
`USE_UART` is by default set to 0 in `inc/uf2.h`. The `#if`'s that use it seem to be OK. What are you changing that causes the errors above? I'm a little confused.

wallarug, 2019-10-08:
I'm going from this config:

```
#ifndef BOARD_CONFIG_H
#define BOARD_CONFIG_H

#define CRYSTALLESS    1

#define VENDOR_NAME "Robotics Masters"
#define PRODUCT_NAME "Robo HAT MM1 M4"
#define VOLUME_LABEL "ROBOM4BOOT"
#define INDEX_URL "https://roboticsmasters.co"
#define BOARD_ID "SAMD51G19A-RoboHATMM1-v24"

#define USB_VID 0x1209
#define USB_PID 0x4D44

#define LED_PIN PIN_PB22

#define BOOT_USART_MODULE                 SERCOM1
#define BOOT_USART_MASK                   APBAMASK
#define BOOT_USART_BUS_CLOCK_INDEX        MCLK_APBAMASK_SERCOM1
#define BOOT_USART_PAD_SETTINGS           UART_RX_PAD1_TX_PAD0
#define BOOT_USART_PAD3                   PINMUX_UNUSED
#define BOOT_USART_PAD2                   PINMUX_UNUSED
#define BOOT_USART_PAD1                   PINMUX_PA17C_SERCOM1_PAD1
#define BOOT_USART_PAD0                   PINMUX_PA16C_SERCOM1_PAD0
#define BOOT_GCLK_ID_CORE                 SERCOM1_GCLK_ID_CORE
#define BOOT_GCLK_ID_SLOW                 SERCOM1_GCLK_ID_SLOW

#endif
```

To this config:

```
#ifndef BOARD_CONFIG_H
#define BOARD_CONFIG_H

#define CRYSTALLESS    1

#define VENDOR_NAME "Robotics Masters"
#define PRODUCT_NAME "Robo HAT MM1 M4"
#define VOLUME_LABEL "ROBOM4BOOT"
#define INDEX_URL "https://roboticsmasters.co"
#define BOARD_ID "SAMD51G19A-RoboHATMM1-v24"

#define USB_VID 0x1209
#define USB_PID 0x4D44

#define LED_PIN PIN_PB22
#endif
```

Nothing too spectacular 😄 .

wallarug, 2019-10-08:
It's easy to fix to make it compile again... it's just a question of is it providing the correct functionality.  It would appear that UART was always building in anyway for all bootloaders but just not running (depending on processor and configuration).

https://github.com/microsoft/uf2-samdx1/blob/master/src/usart_sam_ba.c#L120

```
    #ifdef SAMD51
    #if USE_UART
    GCLK->PCHCTRL[BOOT_GCLK_ID_CORE].reg = GCLK_PCHCTRL_GEN_GCLK0_Val | (1 << GCLK_PCHCTRL_CHEN_Pos);
    GCLK->PCHCTRL[BOOT_GCLK_ID_SLOW].reg = GCLK_PCHCTRL_GEN_GCLK3_Val | (1 << GCLK_PCHCTRL_CHEN_Pos);

    MCLK->BOOT_USART_MASK.reg |= BOOT_USART_BUS_CLOCK_INDEX ;
    #endif
    #endif
```

closed 2020-01-14 by wallarug