microsoft/uf2-samdx1 #79: FAT32

gotfredsen, 2019-10-22:
Can I suggest an upgrade to the bootloader allowing the device to **be seen as a FAT32 device** instead of FAT16.
iOS13 allows to see FAT32 devices, but not FAT16. If it was possible to change to FAT32, it would probably be possible to program a device using the USB connection on an iPhone or a iPad (with a OTG cable).

I have tried changing quite a bit in the `fat.c` file (mostly expanding the `FAT_BootBlock` to include FAT32 fields) as well as increasing the `NUM_FAT_BLOCKS` to 65526 in the `uf2.h`, but I think I am in over my head.

This functionality could be controlled by a define in the `uf2.h` file allowing the user to choose between FAT16 and FAT32.

dhalbert, 2019-10-22:
For CircuitPython, we are successfully presenting a FAT12 filesystem to iOS 13 as CIRCUITPY. See https://learn.adafruit.com/use-circuitpython-devices-with-iphone-ipad. I don't believe the issue with the UF2 drive not appearing is due to it being FAT16 per se: it's some other issue about the presentation of the drive. We are working on this. There is an issue about how much power a board says it draws when it first enumerates.

gotfredsen, 2019-10-23:
Thanks @dhalbert. If work is already in progress, I will close this issue. Thanks again.

closed 2019-10-23 by gotfredsen