microsoft/uf2-samdx1 #80: SAMD SDU library

gotfredsen, 2019-10-22:
Can I suggest an option in the bootloader to choose to have the SAMD SDU library included in the bootloader. Possibly to choose whether to have or not using a `#define` in the `uf2.h` file. 

I have tried to include the 4000 bytes from [SAMD SDU library](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries/SDU) in multiple places, but without luck. I tried to change the `.ld` files to include the `.sketch_boot `section and adding the file, but I believe that I need to change some address sizes from 2k to 6k (for SAMD21) to get it to work, and I am not sure I get all the right places changed.

Today we just add the above library in Arduino, but if we could have it in the bootloader, it would not be forgotten. We have the problem that our software cannot be updated if the previously used software didn't use the library.

dhalbert, 2019-10-22:
Could you explain the scenario which prevents your updating your software without the library? Could you use a two-stage update? The first would load an updater which knows about the SD cards, and the second would run that updater for the final upload.

dhalbert, 2019-10-22:
Adafruit has considered adding the capability to write to external flash chips to UF2, but it could also be an on-board SD card. For SAMD21 boards, though, the bootloader will no longer fit in 8KiB, and we really need the internal flash space on the SAMD21 boards.