microsoft/uf2-samdx1 #90: purpose of flash_write() function in flash_sam21.c?

dhalbert, 2020-04-09:
Hi, I'm trying to gain back some space in the samd21 version of the bootloader to eventually add fuse repair and other features.

Could I ask, what is the purpose of this `flash_write()` function?
https://github.com/microsoft/uf2-samdx1/blob/df89a1f91a7464a51cd11cb3ad7ca10051028bc4/src/flash_samd21.c#L62
It's not used in C code but is mentioned in the `Makefile`:
https://github.com/microsoft/uf2-samdx1/blob/df89a1f91a7464a51cd11cb3ad7ca10051028bc4/Makefile#L197-L198

A samd51 version of this function was never written.

If its purpose has passed, it could be removed to reclaim some space. Thanks.

mmoskal, 2020-04-09:
It was used to generate assembly stub to be uploaded via debugger to flash on device. While you can remove it, it will not save space. We compile with `-ffunction-sections` (wasn't you who added that? ;-) so functions that are not used are not compiled in.

dhalbert, 2020-04-09:
Ah, ok, thanks!

closed 2020-04-09 by dhalbert