microsoft/uf2-samdx1 #91: [Question] Extending storage space with external flash

modmatic, 2020-04-24:
Is it possible to build a UF2 bootloader that can utilize external flash (for instance, QSPI flash on a SAMD51) for mass storage, rather than the built in flash? I believe this is how CircuitPython bootloaders behave. Being able to easily populate the external flash via drag and drop would be very beneficial for boards that include one. I’d expect that the internal flash would still be used for storing/updating the bootloader itself (and perhaps its auxiliary metadata files), but any other types of files (picture assets, JSON metadata, etc) could be stored on the external flash (which would probably need to be formatted to something like FAT).

dhalbert, 2020-04-24:
Hi, just to be clear, CircuitPython itself is not a bootloader. It is "just another program". The CIRCUITPY drive is a real USB drive, which you can populate like any other drive. There is also an Arduino library, https://github.com/adafruit/Adafruit_TinyUSB_Arduino, which can provide USB-drive-style access to external flash. So once you load CircuitPython or an Arduino program that uses that library, you'd have access to external flash.

modmatic, 2020-04-24:
@dhalbert Ah that makes much more sense, thanks! The main goal is to allow easy access to the external flash for users, so I’ll look into TinyUSB. Having the ability via a UF2 bootloader, so the can be there regardless of which program is installed, would be even better, but obviously is a more complex solution.

closed 2020-04-24 by dhalbert

modmatic, 2020-05-09:
Not intending to reopen, but I have a lingering question related to your comment:

This seems to mean there is a way to upload programs via drag-and-dropping UF2 files. Is there a documented process for converting Arduino programs to UF2? (For instance, from .hex; or, even better, compile directly to UF2? My first instinct would be to grab the current UF2 file from a programmed chip and see if that file contains the whole flash contents, but I haven’t tested that yet.

mmoskal, 2020-05-09:
This script https://github.com/microsoft/uf2/blob/master/utils/uf2conv.py can convert hex and bin to UF2. The CURRENT.UF2 does contain the whole flash.

modmatic, 2020-05-10:
@mmoskal Thanks for that! Good to know about the current UF2 containing the whole flash. We may opt to ship chips with a program pre-loaded (along with the bootloader), so that would be perfect.