microsoft/uf2-samdx1 #93: UF2 MSC volume does not show up under Linux - MacOS works fine

spachner, 2020-06-06:
Hi,

I found your uf2 bootloader which looks quite promising. I have a custom board with SAMD21 which works file with USB/MSC driver from Microchip asf. Now I would like to add a USB bootloader and trying your uf2 (msc) to get drag-and-drop firmware update work. I develop on MacOS and there everything work quite fine. When attaching my SAMD board the new volume shows up and firmware update works perfectly!
Unfortunately I would like to update the firmware under Linux (Ubuntu). When attaching my board almost never a volume shows up. Despite the dmesg look quite good (see below). I rare cases a volume did show up, so it does not seem impossible under Linux. I do not understand the difference between MacOS and Linux. Why is MacOS working and Linux almost never?
Seems that the device is recognized as USB MSC but the registration does not get through  and finally not drive device (/dev/sd?) is created and finally no volume with filesystem is accessable.

What do I miss?

regards

spachner

```
[Sa Jun  6 19:48:19 2020] usb 2-2: new full-speed USB device number 3 using uhci_hcd
[Sa Jun  6 19:48:20 2020] usb 2-2: New USB device found, idVendor=203a, idProduct=fffe, bcdDevice= 1.01
[Sa Jun  6 19:48:20 2020] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sa Jun  6 19:48:20 2020] usb 2-2: Product: Virtual USB1.1 HUB
[Sa Jun  6 19:48:20 2020] usb 2-2: Manufacturer: Parallels
[Sa Jun  6 19:48:20 2020] usb 2-2: SerialNumber: PW3.0
[Sa Jun  6 19:48:20 2020] hub 2-2:1.0: USB hub found
[Sa Jun  6 19:48:20 2020] hub 2-2:1.0: 15 ports detected
[Sa Jun  6 19:48:20 2020] usb 2-2.11: new full-speed USB device number 4 using uhci_hcd
[Sa Jun  6 19:48:20 2020] usb 2-2.11: New USB device found, idVendor=03eb, idProduct=2402, bcdDevice=42.01
[Sa Jun  6 19:48:20 2020] usb 2-2.11: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sa Jun  6 19:48:20 2020] usb 2-2.11: Product: SafeTag
[Sa Jun  6 19:48:20 2020] usb 2-2.11: Manufacturer: Kinexon GmbH
[Sa Jun  6 19:48:20 2020] usb 2-2.11: SerialNumber: EE1506825053334D332E3120FF081B3F
[Sa Jun  6 19:48:20 2020] hid-generic 0003:03EB:2402.0003: hiddev4,hidraw2: USB HID v1.00 Device [Kinexon GmbH SafeTag] on usb-0000:00:1d.0-2.11/input3
[Sa Jun  6 19:48:20 2020] cdc_acm 2-2.11:1.0: ttyACM0: USB ACM device
[Sa Jun  6 19:48:20 2020] usbcore: registered new interface driver cdc_acm
[Sa Jun  6 19:48:20 2020] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[Sa Jun  6 19:48:20 2020] usb-storage 2-2.11:1.2: USB Mass Storage device detected
[Sa Jun  6 19:48:20 2020] scsi host8: usb-storage 2-2.11:1.2
[Sa Jun  6 19:48:20 2020] usbcore: registered new interface driver usb-storage
[Sa Jun  6 19:48:20 2020] usbcore: registered new interface driver uas
[Sa Jun  6 19:48:42 2020] usb 2-2.11: reset full-speed USB device number 4 using uhci_hcd
[Sa Jun  6 19:48:42 2020] cdc_acm 2-2.11:1.0: ttyACM0: USB ACM device
```

dhalbert, 2020-06-06:
This does work on Linux; I use it all the time. I notice you are running Linux on Parallels. Do you have a native Linux machine to try?

spachner, 2020-06-06:
Hi Dan,

thank for your quick answer. Glad to hear that is must work. I am working with Linux@Parallels for many years now and never had an issue with USB. I will try on native Linux next week.
I also gave CDC a try and used bossac. 
Either on MacOS and on Linux@Parallels. Same result Mac works and Linux not.
I guess the CDC device is /dev/ttyACM0. I tried

```
[linux]$
bossac -i -p /dev/ttyACM0
No device found on /dev/ttyACM0
```

```
[macOS]$
[ master ] $ ./bin/bossac -i -p /dev/tty.usbmodem1412201
Device       : ATSAMD21x18
Version      : v1.1 [Arduino:XYZ] Jun  6 2020 18:42:39
Address      : 0x0
Pages        : 4096
Page Size    : 64 bytes
Total Size   : 256KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
BOD          : false
BOR          : false
```

I have no glue what is going on. My app running also MSC on my board which works fine under Linux@Parallels.

I will come back to you next week.

spachner

spachner, 2020-06-08:
Hi Dan,

quite wired: After rebooting the Mac my virtual Ubuntu works quite well with uf2. I am happily bootloading now. Seems to be a glitch with MacOS's USB subsystem.

Thanks for you help.

regards

spachner

closed 2020-06-08 by spachner

dhalbert, 2020-06-08:
Glad to hear! I was thinking of how to reproduce this.

presslab-us, 2020-07-31:
Just thought I'd chime in and say that I see the same problem where it resets after 22 seconds, using USB3 redirection with QEMU 5.0.0.  On native Linux it is working fine.

I'd guess it's some kind of timing issue but who knows.