microsoft/uf2-samdx1 #97: I2C in the bootloader

CatalinPopovici, 2021-03-16:
Hello, I am trying to implement Bitbang I2C protocol in the bootloader. I was not able to find any code that would work for any bootloader. After that, I have found an Arduino Bitbang I2C library which I have successfully compiled to the bootloader using Arduino IDE and was able to use the lib with Arduino IDE code but when I tried to implement the source code in the bootloader, I realized that it uses Linux API’s (which the bootloader compiler does not support)

dhalbert, 2021-03-16:
Any low-level library like this will vary by the chip family. Which chip family are you targeting?

Is this the library you are using? https://www.arduino.cc/reference/en/libraries/bitbang_i2c/. It supports AVR-style chips, but you are probably targeting something else.

CatalinPopovici, 2021-03-16:
Hello Dan, that is indeed the library I was trying to implement. I am working with Adafruit ItsyBitsy M4 Express.