microsoft/uf2-samdx1 #100: bin2uf2.js throwing deprecation warning

sslupsky, 2021-06-10:
When running the script bin2uf2.js, a deprecation warning is displayed in terminal:
```
(node:33441) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
(Use `node --trace-deprecation ...` to show where the warning was created)
```

Here is the deprecation trace:
```
(node:33486) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
    at showFlaggedDeprecation (node:buffer:184:11)
    at new Buffer (node:buffer:268:3)
    at Object.<anonymous> (/Users/stevenslupsky/Library/Mobile Documents/com~apple~CloudDocs/source/uf2-samdx1/scripts/bin2uf2.js:16:14)
    at Module._compile (node:internal/modules/cjs/loader:1109:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1138:10)
    at Module.load (node:internal/modules/cjs/loader:989:32)
    at Function.Module._load (node:internal/modules/cjs/loader:829:14)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:76:12)
    at node:internal/main/run_main_module:17:47
```

I am running macOS Big Sur.
Node version 16.3.0

mmoskal, 2021-06-10:
Thanks, should be fixed now!

closed 2021-06-10 by mmoskal

sslupsky, 2021-06-10:
wow, that was fast!  Thank you :-)

commit be9a2c4d (2021-06-10): Use Buffer.alloc() - fixes #100