microsoft/uf2-samdx1 #102: [Compile Error] ATSAMD21J17A

kelroy1990, 2021-08-13:
Hi, i am just working on a new ATSAMD21J17A from Microchip, as you know, its really complex to obtain a ATSAMD21G18 at this moment, so i am trying to make my proyect with the J17A variant.

The problem comes when i try to compile the new board, i obtain the following error:

`Building zero
echo "src/flash_samd21.c"
src/flash_samd21.c
arm-none-eabi-gcc -mthumb -mcpu=cortex-m0plus -Os -g -DSAMD21 -x c -c -pipe -nostdlib --param max-inline-insns-single=500 -fno-strict-aliasing -fdata-sections -ffunction-sections -D__SAMD21J17A__ -Werror -Wall -Wstrict-prototypes -Werror-implicit-function-declaration -Wpointer-arith -std=gnu99 -ffunction-sections -fdata-sections -Wchar-subscripts -Wcomment -Wformat=2 -Wimplicit-int -Wmain -Wparentheses -Wsequence-point -Wreturn-type -Wswitch -Wtrigraphs -Wunused -Wuninitialized -Wunknown-pragmas -Wfloat-equal -Wno-undef -Wbad-function-cast -Wwrite-strings -Waggregate-return -Wformat -Wmissing-format-attribute -Wno-deprecated-declarations -Wpacked -Wredundant-decls -Wnested-externs -Wlong-long -Wunreachable-code -Wcast-align -Wno-missing-braces -Wno-overflow -Wno-shadow -Wno-attributes -Wno-packed -Wno-pointer-sign  -I. -I./inc -I./inc/preprocessor -I./boards/zero -Ilib/cmsis/CMSIS/Include -Ilib/usb_msc -Ibuild/zero -Ilib/samd21/samd21a/include/ src/flash_samd21.c -o build/zero/flash_samd21.o
In file included from src/flash_samd21.c:1:0:
./inc/uf2.h:293:73: error: division by zero [-Werror=div-by-zero]
 #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
                                                                         ^
./inc/uf2.h:298:1: note: in expansion of macro 'STATIC_ASSERT'
 STATIC_ASSERT(FLASH_NUM_ROWS * 4 == FLASH_NB_OF_PAGES);
 ^~~~~~~~~~~~~
./inc/uf2.h:293:42: error: enumerator value for '_static_assert_298' is not an integer constant
 #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
                                          ^
./inc/uf2.h:291:24: note: in definition of macro 'CONCAT_1'
 #define CONCAT_1(a, b) a##b
                        ^
./inc/uf2.h:293:33: note: in expansion of macro 'CONCAT_0'
 #define STATIC_ASSERT(e) enum { CONCAT_0(_static_assert_, __LINE__) = 1 / ((e) ? 1 : 0) }
                                 ^~~~~~~~
./inc/uf2.h:298:1: note: in expansion of macro 'STATIC_ASSERT'
 STATIC_ASSERT(FLASH_NUM_ROWS * 4 == FLASH_NB_OF_PAGES);
 ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
Makefile:159: fallo en las instrucciones para el objetivo 'build/zero/flash_samd21.o'
make: *** [build/zero/flash_samd21.o] Error 1
`

I am trying to understand the error, but my knowledge with that kind of macros its really low.

Any help could be appreciated. Thanks.

dstahlke, 2021-12-21:
I got this same error.  Attempted to fix it by editing inc/uf2.h:

    -#define FLASH_NUM_ROWS 1024
    +#define FLASH_NUM_ROWS 512

That allows it to get past the STATIC_ASSERT, but then there is a linker error:

    /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: section .binfo LMA [0000000000001ff0,0000000000001fff] overlaps section .relocate LMA [0000000000001e30,0000000000001ffb]

dstahlke, 2021-12-22:
I don't mean to hijack the thread with my own issues, but I did get past the linker error by turning off some features in inc/uf2.h, namely USE_FAT, USE_INDEX_HTM, and USE_WEBUSB.  Probably these features just made the bootloader too big for this chip.

Unfortunately, I'm now fighting OpenOCD issues.  So I can't yet report on whether these fixes work.

kelroy1990, 2021-12-28:
@dstahlke 

I finally manage to solve this issue, i just change the chip compilation for the ATSAMD21G17, its a bit tricky but it works.

At this moment i dont have time to get the diff to see what its the real problem right now, but it should be a define or something like that.

Regards, Carlos.

dstahlke, 2021-12-28:
For me, it was enough to do the FLASH_NUM_ROWS change I mentioned above, set #define USE_INDEX_HTM 0 to make it a bit smaller, and define the chip type in board.mk:

CHIP_FAMILY = samd21
CHIP_VARIANT = SAMD21J17A

Also, -Wno-deprecated in the top Makefile and editing scripts/dbgtool.js to use a patched OpenOCD with ST-Link (master branch and releases of OpenOCD have a bug when using ST-Link with this chip).

With that, bossac in zephyr doesn't work, but putting *.uf2 on the USB FAT works.