microsoft/uf2-samdx1 #115: [compile error] Arduino Zero

deladriere, 2024-08-11:
I am running the script with 
```
make BOARD=zero
```
everything run fines until it reaches 
```
cc1: fatal error: build/zero/selfdata.c: No such file or directory
compilation terminated
```

any ideas to solve this ?

dhalbert, 2024-08-11:
`selfdata.c` is generated by `scripts/gendata.py`. When I run the `make` command above on a fresh clone, the build fails because is there is no `python2`. Are you seeing that as well?

I maintain https://github.com/adafruit/uf2-samdx1. It is more up-to-date in some respects; we work closely with upstream. I don't see such an error there.

deladriere, 2024-08-11:
Woaw thanks for the update !
Sorry, I should have mentioned that I corrected the script to use Python 3 instead of Python 2, but I'm still encountering the error I mentioned earlier. 
in fact, I came here because I was using the Adafruit version:  UF2 dragging works fine, but as I code with PlatformIO, I prefer to upload via the emulated serial port. However, this often fails and stops midway  (I'm working with my own SAMD21 board), so I plan to compare both (Adafruit and Microsoft). (btw, the classical Arduino bootloader never fails to upload)